joostn/OpenJsCad #65: Signiture of setColor() is inconsist

z3dev, 2015-05-11:
The newest version of CSG.js has introduced inconsistency in CSG.prototype.setColor(r,g,b,a) and CSG.Polygon.prototype([r,g,b,a])

As for the correct signature, I suggest supporting setColor([r,g,b,a]) as the provides the best performance, i.e. the array can be stored directly.

We (CSG.js users) can easily convert from r,g,b,a to [r,g,b,a] as well.

bebbi, 2015-05-12:
Tx, I'll look into this. not sure I get the performance argument. Do you plan to submit additional bugs so I can look at it when we have them together?

Spiritdude, 2015-05-13:
I recommend to permit both: setColor(r,g,b(,a)) and setColor([r,g,b(,a)])

z3dev, 2015-05-13:
Please address this issue first.

The CSG.Shared object retains an array. So, passing the color specification by array will improve performance. Although, permitting both styles would probably be the best for the users of CSG.js.

bebbi, 2015-05-15:
@z3dev that's an easy one, can you write a pull request against gh-pages? I'd tend to support the version that's already documented but I'm fine with supporting both for both setColor functions. If I understood you well, I don't believe there's a performance difference in creating an array inside a function call vs inside a function body.

z3dev, 2015-05-19:
Hello. I don't have a patch for this.

Please correct and let me know when the new version is available.

joostn, 2015-05-19:
See https://github.com/joostn/OpenJsCad/pull/66
I'm proposing to use the array syntax from now on, with the previous syntax supported for backwards compatibility, I don't think performance is relevant here.

closed 2015-05-19 by bebbi