joostn/OpenJsCad #71: Union behaving as a subtract with polyhedron

aurelient, 2015-08-06:
I've a somewhat complex polyhedron : a smartphone case converted from STL.

I can display it nicely, but operations with basic shapes (Cube, Sphere) are not behaving as they should.

For example union seems to operate as a subtract.

Pasting the code in the first comment

aurelient, 2015-08-06:
function main() {

var phoneCase = CSG.polyhedron({
 points:[[-30.6751,53.245,0],[7.07935,47.5656,0],[-30.6751,-53.245,0],[7.07935,47.5656,0],[-30.6751,53.245,0],[5.94585,48.4958,0],[-30.6751,-53.245,0],[-22.4991,-63.1196,0],[-24.4069,-62.5408,0],[5.94585,48.4958,0],[-30.6751,53.245,0],[-20.615,63.3051,0],[-30.6751,-53.245,0],[-24.4069,-62.5408,0],[-26.1651,-61.601,0],[-20.615,63.3051,0],[-30.6751,53.245,0],[-22.4991,63.1196,0],[-30.6751,-53.245,0],[-26.1651,-61.601,0],[-27.7063,-60.3363,0],[-22.4991,63.1196,0],[-30.6751,53.245,0],[-24.4069,62.5408,0],[-30.6751,-53.245,0],[-27.7063,-60.3363,0],[-28.971,-58.7951,0],[-24.4069,62.5408,0],[-30.6751,53.245,0],[-26.1651,61.601,0],[-30.6751,-53.245,0],[-28.971,-58.7951,0],[-29.9108,-57.0369,0],[-26.1651,61.601,0],[-30.6751,53.245,0],[-27.7063,60.3363,0],[-30.6751,-53.245,0],[-29.9108,-57.0369,0],[-30.4896,-55.1291,0],[-27.7063,60.3363,0],[-30.6751,53.245,0],[-28.971,58.7951,0],[-28.971,58.7951,0],[-30.6751,53.245,0],[-29.9108,57.0369,0],[-29.9108,57.0369,0],[-30.6751,53.245,0],[-30.4896,55.1291,0],[-30.685,-53.245,0],[-30.6751,53.245,0],[-30.6751,-53.245,0],[-30.6751,53.245,0],[-30.685,-53.245,0],[-30.685,53.245,0],[-20.615,-63.3051,0],[20.615,-63.3051,0],[20.615,-63.315,0],[20.615,-63.3051,0],[11.235,46.305,0],[19.325,46.305,0],[20.615,-63.3051,0],[9.77573,46.4487,0],[11.235,46.305,0],[-30.6751,-53.245,0],[20.615,-63.3051,0],[-20.615,-63.3051,0],[20.615,-63.3051,0],[-30.6751,-53.245,0],[9.77573,46.4487,0],[-30.6751,-53.245,0],[-20.615,-63.3051,0],[-22.4991,-63.1196,0],[9.77573,46.4487,0],[-30.6751,-53.245,0],[8.37254,46.8744,0],[8.37254,46.8744,0],[-30.6751,-53.245,0],[7.07935,47.5656,0],[5.94585,48.4958,0],[-20.615,63.3051,0],[5.01561,49.6293,0],[4.32439,50.9225,0],[-20.615,63.3051,0],[3.89873,52.3257,0],[5.01561,49.6293,0],[-20.615,63.3051,0],[4.32439,50.9225,0],[-20.615,-63.3051,0],[20.615,-63.315,0],[-20.615,-63.315,0],[20.7843,46.4487,0],[30.685,-53.245,0],[30.6751,-53.245,0],[30.685,-53.245,0],[24.6142,48.4958,0],[30.685,53.245,0],[20.615,-63.3051,0],[30.6751,-53.245,0],[30.4896,-55.1291,0],[25.5444,49.6293,0],[30.685,53.245,0],[24.6142,48.4958,0],[20.615,-63.3051,0],[30.4896,-55.1291,0],[29.9108,-57.0369,0],[30.685,53.245,0],[26.2356,50.9225,0],[30.6751,53.245,0],[20.615,-63.3051,0],[29.9108,-57.0369,0],[28.971,-58.7951,0],[26.6613,52.3257,0],[30.6751,53.245,0],[26.2356,50.9225,0],[20.615,-63.3051,0],[28.971,-58.7951,0],[27.7063,-60.3363,0],[26.805,53.885,0],[30.6751,53.245,0],[26.805,53.785,0],[26.6613,55.3443,0],[30.4896,55.1291,0],[26.805,53.885,0],[30.4896,55.1291,0],[26.6613,55.3443,0],[29.9108,57.0369,0],[30.6751,53.245,0],[26.805,53.885,0],[30.4896,55.1291,0],[20.615,-63.3051,0],[27.7063,-60.3363,0],[26.1651,-61.601,0],[26.805,53.785,0],[30.6751,53.245,0],[26.6613,52.3257,0],[26.2356,50.9225,0],[30.685,53.245,0],[25.5444,49.6293,0],[20.615,-63.3051,0],[26.1651,-61.601,0],[24.4069,-62.5408,0],[30.685,-53.245,0],[23.4807,47.5656,0],[24.6142,48.4958,0],[20.615,-63.3051,0],[24.4069,-62.5408,0],[22.4991,-63.1196,0],[30.685,-53.245,0],[22.1875,46.8744,0],[23.4807,47.5656,0],[30.685,-53.245,0],[20.7843,46.4487,0],[22.1875,46.8744,0],[30.6751,-53.245,0],[20.615,-63.3051,0],[19.325,46.305,0],[30.6751,-53.245,0],[19.325,46.305,0],[20.7843,46.4487,0],[26.2356,56.7474,0],[29.9108,57.0369,0],[26.6613,55.3443,0],[29.9108,57.0369,0],[26.2356,56.7474,0],[28.971,58.7951,0],[25.5444,58.0406,0],[28.971,58.7951,0],[26.2356,56.7474,0],[28.971,58.7951,0],[25.5444,58.0406,0],[27.7063,60.3363,0],[25.0225,58.6765,0],[27.7063,60.3363,0],[25.5444,58.0406,0],[27.7063,60.3363,0],[25.0225,58.6765,0],[26.1651,61.601,0],[24.4984,59.1066,0],[26.1651,61.601,0],[25.0225,58.6765,0],[24.4984,59.1066,0],[24.4069,62.5408,0],[26.1651,61.601,0],[23.2588,59.7692,0],[24.4069,62.5408,0],[24.4984,59.1066,0],[23.2588,59.7692,0],[22.4991,63.1196,0],[24.4069,62.5408,0],[21.9138,60.1772,0],[22.4991,63.1196,0],[23.2588,59.7692,0],[21.9138,60.1772,0],[20.615,63.3051,0],[22.4991,63.1196,0],[20.515,60.315,0],[20.615,63.3051,0],[21.9138,60.1772,0],[7.47341,60.315,0],[20.615,63.3051,0],[20.515,60.315,0],[20.615,63.3051,0],[7.47341,60.315,0],[20.615,63.315,0],[3.89873,52.3257,0],[-20.615,63.3051,0],[3.755,53.785,0],[3.755,53.785,0],[-20.615,63.3051,0],[3.755,53.885,0],[3.755,53.885,0],[-20.615,63.3051,0],[3.89873,55.3443,0],[-20.615,63.3051,0],[4.32439,56.7474,0],[3.89873,55.3443,0],[-20.615,63.315,0],[4.32439,56.7474,0],[-20.615,63.3051,0],[4.32439,56.7474,0],[-20.615,63.315,0],[5.01561,58.0406,0],[5.01561,58.0406,0],[-20.615,63.315,0],[5.94585,59.1741,0],[5.94585,59.1741,0],[-20.615,63.315,0],[7.07935,60.1044,0],[7.07935,60.1044,0],[-20.615,63.315,0],[7.47341,60.315,0],[7.47341,60.315,0],[-20.615,63.315,0],[20.615,63.315,0],[29.285,53.145,0.799987],[29.685,53.145,0.799988],[29.285,53.145,0.799988],[29.685,53.145,0.799988],[29.285,53.145,0.799987],[29.685,53.145,0.799987],[29.685,53.145,0.799988],[29.685,53.145,0.799987],[29.685,53.145,7.25],[22.6161,63.708,9.45], [22.5,63.7195,4.625],[22.5,63.7195,9.45],[22.6161,63.708,0.599991],[22.5,63.7195,4.625],[22.6161,63.708,9.45],[22.6161,63.708,0.599991],[22.4265,63.7267,3.87879],[22.5,63.7195,4.625],[22.6161,63.708,0.599991],[22.2088,63.7482,3.16124],[22.4265,63.7267,3.87879],[22.6161,63.708,0.599991],[21.8553,63.783,2.49996],[22.2088,63.7482,3.16124],[22.6161,63.708,0.599991],[21.3797,63.8298,1.92032],[21.8553,63.783,2.49996],[22.6161,63.708,0.599991],[20.8,63.8869,1.44464],[21.3797,63.8298,1.92032],[20.615,63.9051,0.599991],[20.8,63.8869,1.44464], [... cut ...] 4150,4151],[4152,4153,4154],[4155,4156,4157],[4158,4159,4160],[4161,4162,4163],[4164,4165,4166],[4167,4168,4169],[4170,4171,4172],[4173,4174,4175],[4176,4177,4178],[4179,4180,4181],[4182,4183,4184],[4185,4186,4187],[4188,4189,4190],[4191,4192,4193],[4194,4195,4196],[4197,4198,4199],[4200,4201,4202],[4203,4204,4205],[4206,4207,4208],[4209,4210,4211],[4212,4213,4214],[4215,4216,4217]]});

// a cube:
var cube = CSG.cube({
  center: [0, 0, 0],
  radius: [5, 5, 5]
});

return phoneCase.union(cube);
}

aurelient, 2015-08-06:
Here is the result from the same operation with openscad :
[image]

bebbi, 2015-08-13:
This is because your face normals are inwards out from openjscad perspective.
Try this to confirm:

    var faces = <your array of faces>;
    faces = faces.map(function(face) {
        return [face[1], face[0], face[2]];
    });
    var phoneCase = CSG.polyhedron({
        points: <your array of points>,
        faces: faces
    });

Can you try the changed orientation in OpenSCAD as well? Does it have same or different behaviour?
Is the conclusion that openjscad syntax is the same, or opposite, as OpenSCAD?

aurelient, 2015-08-13:
Thanks for the input. Indeed changing the faces fixed the problem.

In OpenSCAD, it leads to the same result (before and after changing faces) : a union as in the picture above https://github.com/joostn/OpenJsCad/issues/71#issuecomment-128351960.

v1 (face normals inwards) :
- openjscad subtract == openSCAD union
- openjscad union == openSCAD subtract

v2 (as you suggested)
- openjscad union == openSCAD union

Thanks again, and congrats to @joostn and all the contributors for such a great library.

I let you decide whether the issue should be closed or not, as I don't know how close you want to be to openSCAD

bebbi, 2015-08-13:
> In OpenSCAD, it leads to the same result (before and after changing faces) : a union as in the picture above #71 (comment).

Thanks Aurélien for raising this.
@joostn When I defined polyhedron, for some reason I got the impression that OpenSCAD is defining inward normals against all conventions, and adapted the signature with this comment:
// openscad convention defines inward normals - so we have to invert here
However, from OpenSCAD manual: When looking at the face from the outside inwards, the points must be clockwise.

I suggest to just reverse the face orientation in polyhedron even if it may break some stuff.
Thoughts?

joostn, 2015-08-13:
Hi Stefan,
I agree, let's make sure it's compatible with OpenSCAD's polyhedron.

bebbi, 2015-08-13:
Ok I'll change it

bebbi, 2015-08-14:
Sorry, false alarm. The openjscad spec is already in line with the OpenJSCAD spec, no change needed.

It's just that OpenSCAD doesn't seems to objects and their inversions the same, where openjscad interprets them as the infinite complements of the defined objects.
So, @aurelient s STL was inverted but OpenSCAD apparently still worked.
Going to close this issue.
@aurelient , maybe helpful if you can mention the process used for converting an STL to a list of points/faces here - in case others run into the same issue

closed 2015-08-14 by bebbi

aurelient, 2015-08-15:
Hi, I used the following javascript converter :
http://www.thingiverse.com/thing:62666/#instructions
The converter above was also ported to python.