joostn/OpenJsCad #72: Feature Request: RoundRadius on CSG.roundedCylinder()

z3dev, 2015-10-01:
This is a very useful function to make rounded cylinders of a given radius, but the rounded end of the cylinder should also be configurable. Currently, the rounded end of the cylinder is the same as the radius of the cylinder. But anything between 0 > rounded end radius <= radius is possible.

Parameters:
    //   start: start point of cylinder (default [0, -1, 0])
    //   end: end point of cylinder (default [0, 1, 0])
    //   radius: radius of cylinder (default 1), must be a scalar
    //   resolution: determines the number of polygons per 360 degree revolution (default 12)
    //   normal: a vector determining the starting angle for tesselation. Should be non-parallel to start.minus(end)

Add...
// roundedRadius: radius of rounded end of cylinder, default radius of cylinder

[image]

bebbi, 2015-10-01:
That would be a reasonable addition. I don't have the time now, but do you want to take a stab at it? I would probably go about it by replacing the existing logic with something like this:
1. create a path, using arc, with all params (radius, roundradius, height)
2. convert to CAG using path.innerToCAG()
3. rotateExtrude to CSG
This way, if you want you could even expose the angle param of rotateExtrude as a sectorAngle argument for rounded cylinder pies.

z3dev, 2015-10-02:
@bebbi thanks
My version was very similar, but I used solidFromSlices(). It's just a matter of following the points, and scaling the circle.

FYI, there seem to be issues with using rotateExtrude with a CAG that has a zero(0) X offset. The internal vertices seemed to be confused.

bebbi, 2015-10-02:
From http://joostn.github.io/OpenJsCad/ docu:
// Note: for this to work, shape shouldn't have point in negative x space
If the CAG is valid and all x>=0 overall, it should work - if not, maybe you can open an issue with the example?

Again, please feel free to post PRs to csg.js I’m tight on time right now, so any help is appreciated.

z3dev, 2017-02-26:
This issue was moved to jscad/csg.js#21

closed 2017-02-26 by z3dev