joostn/OpenJsCad #74: Feature Request: Correct closure mistakes in CSG

z3dev, 2015-11-02:
I came across an issue while writing the conversion routine for SVG to CSG. SVG supports the concept of 'closure" via the 'z' command in PATH elements. The 'z' command functionality is the same as CSG.close().

However, SVG also supports 'manual' closure of paths by joining the last line segment to the starting point. This is not possible in CSG.

I think that CSG.appendPoint(), etc routines could check for the 'manual' closure. Or CSG.close() could correct the internal set of points if start and end points are the same.

joostn, 2015-11-02:
Actually I think it's fine if CSG.Path2D can have multiple subsequent points with the same coordinates, and we could also allow to make the 'mistake' of copying the first point at the end of the Path2D AND closing it too (effectively resulting in duplicate points).

Duplicate points do become a problem when converting to CAG (using Path2D.innerToCAG), so I'd propose to modify CAG.fromPoints to check for duplicate points and remove them. Should be easy. And Path2D.innerToCAG could be modified to check if the first and last points match, and if so, also allow a path without the 'closed' flag.

BTW in your import routines you're implicitly converting svg paths to CAG. There may be occasions though where you'd want a Path2D instead, for example to extrude a path to create a 'snake' like feature by tracing it with a sphere. I'd say it should convert to Path2D by default, this would also solve the issue of self intersecting paths: Path2D can be self intersecting; Path2D.innerToCAG (currently) cannot handle self intersecting paths.

z3dev, 2015-11-02:
The code has been changed to produce Path2D paths only. If the path is closed, then innerToCAG() is produced as well. If the path is left open then expandToCAG() is produced.

I like your proposals for Path2D. It sounds like some work on the internals, so I'll leave those changes to you.

Of course, I'll run tests again with the SVG conversion, etc.

z3dev, 2017-02-26:
This issue was moved to jscad/csg.js#22

closed 2017-02-26 by z3dev