joostn/OpenJsCad #81: getFeature: center

dominictarr, 2016-03-15:
It would be great to be able to get the center of a openjscad shape.
That would make this useful to design, say, floating objects, i.e. boats!
You have volume, which is very useful, because you can calculate intersect a boat hull with shape representing the water, and thus calculate the displacement, if you know the center of that displacement, you can compare that to the center of the boat's mass, and know whether it is stable there.

dominictarr, 2016-03-15:
oh, looking closer I see I could center an object, then compare the bounds against an object that was uncentered, this would basically tell me how the center has moved.

bebbi, 2016-03-15:
the center function would give you the geometric center. A getFeatures() function could give you the center of mass if you'd like that. See branch add_centerofmass for a quick hack. If you'd like to make this more elegant, I'm happy to take a contrib!

dominictarr, 2016-03-16:
what is the difference between the geometric center and the center of mass? do you mean the center of mass takes into account the mass of the various parts of the model?

bebbi, 2016-03-16:
Exactly. Imagine a boat with a long mast.
Geometric center will go right up and sit close to the middle of the mast, while center of mass stays low inside the boat. Nb the openjscad impl in that branch assumes same material ('density') used across print.

dominictarr, 2016-03-20:
got it. to calculate the water line i'd intersect the boat model with a plane to represent the water line, water has a uniform density, so if the boat is loaded to a certain weight, it will float at that level. You don't really know the weight of the boat before building it and loading it, so it's useful to know a range... It's also important to know how far over the boat can go and still come back up, so you repeat these calculations on many different angles.

z3dev, 2017-02-26:
This issue was moved to jscad/csg.js#23

closed 2017-02-26 by z3dev