joostn/OpenJsCad #87: fr: minkowski sum

salamanders, 2016-11-18:
This project is awesome.  I stumbled across it after using OpenSCAD and wondering why I had to learn a new language, why couldn't I just use JS, and lo and behold, here this was.  Thank you!  [forum post here.](http://forum.openscad.org/OpenSCAD-and-JavaScript-tt19116.html)

I immediately tried to transfer my script, and couldn't do rounded edges, because no minkowski sum.  It is the best way I know to round out a cube's edges by summing with a sphere.

```javascript
/*
Must be printed with support material.  
Best have support extend outwards so you can yank it out.
*/

$fn = 40;
// my badge dimensions
badge = [54, 87.31, 2.4];

module copy_mirror(vec = [0, 1, 0]) {
  children();
  mirror(vec) children();
}

union() {
    difference() {
        // Main structure
        union() {
            // Rounded edges for the main card holder
            minkowski() {
                cube([badge[0], badge[1], badge[2]-2], center = true);
                sphere(r = 2, center = true);
            };
            // Badge clip extension, 13mm wide
            color("green")
            translate([0, badge[1] / 2, 0])
            minkowski() {
                cube([17, 8, badge[2]-2], center = true);
                sphere(r = 2, center = true);
                //cylinder(r=2,h=1, center = true);
            }
        }
        
        // Bevel cutout replacing badge for angled edges
        color("red")
        cube(badge, center = true);

        // Viewing area cutout
        color("blue")
        cube([badge[0] - 5, badge[1] - 5, 20], center = true);
        
        // Badge loop cutout
        color("purple")
        translate([0, badge[1] / 2 + 1, 0])
        cube([13, 3, 20], center = true);

        // Top insert badge slot
        color("orange")
        translate([0, badge[1] / 2,0])
        cube([badge[0], 40, badge[2]-0.2], center = true);
    };
}
```

joostn, 2016-11-18:
Hi!
There's no generic minkowski sum but you can actually do a minkowski sum with a sphere:
csg.expand() does just that.

See http://joostn.github.io/OpenJsCad/ (section Expansion and Contraction)

z3dev, 2017-02-26:
@salamanders I think this request would be better as part of the scad-api library of the JSCAD organization. Here https://github.com/jscad/scad-api

I can move this issue to that repository if you want. Please let me know.

salamanders, 2017-03-01:
Yes please.

z3dev, 2019-10-11:
Will not fix.

closed 2019-10-11 by z3dev