joostn/OpenJsCad #90: Error "ReferenceError: difference is not defined"

TommyK82, 2017-01-17:
Hi,

I don't know if this is the right place, but here is my question:

I made a JSCAD file and it works at "http://openjscad.org/";  but it does not work at the parser here  (http://joostn.github.io/OpenJsCad/processfile.html)

If I drop it at the dropzone, than the following errors are showing:

```
ReferenceError: difference is not defined
Stack trace:
ReferenceError: difference is not defined
    at main (blob:http://joostn.github.io/0c59e92f-2d3d-4bbf-9fb4-d4ddec4b958f:28:2)
    at Function.OpenJsCad.runMainInWorker (http://joostn.github.io/OpenJsCad/src/openjscad.js:312:18)
    at blob:http://joostn.github.io/0c59e92f-2d3d-4bbf-9fb4-d4ddec4b958f:71:96
```

That is my JSCAD Code:

```
function getParameterDefinitions() {
  return [
	{ name: 'd1', type: 'float', initial: 10, caption: "Insidediameter in mm:" },
	{ name: 'd2', type: 'float', initial: 12, caption: "Outsidediameter in mm:" },
	{ name: 'b1', type: 'float', initial: 10, caption: "Length in mm:" },
	{ name: 'fl', type: 'choice', caption: 'Flange?', values: [0, 1], captions: ["Yes", "No"], initial: 0 },
	{ name: 'd3', type: 'float', initial: 14, caption: "Flangediameter in mm:" },
	{ name: 'b2', type: 'float', initial: 1, caption: "Flangethickness in mm:" },
	{ name: 'sl', type: 'choice', caption: 'Slit?', values: [0, 1], captions: ["Yes", "No"], initial: 0 },
	{ name: 'b3', type: 'float', initial: 1, caption: "Slit-width in mm:" },
	{ name: 'a', type: 'choice',  values: [0, 10, 20], captions: ["0", "10", "20"], initial: 10, caption: "Slit-angle:" }
];
}

function main(params) {

if (params.fl == 1) params.d3 = params.d2;
if (params.fl == 1) params.b2 = 0;
if (params.d1 <= 6) f = 0.3;
if (params.d1 > 6 && params.d1 <= 12) f = 0.5;
if (params.d1 > 12 && params.d1 <= 30) f = 0.8;
if (params.d1 > 30) f = 1.2;
if (params.sl == 1) {e = params.b1;} else {e = 0;}

   return [

	difference(
		union(
			cylinder({d: params.d2, h: params.b1-f, fn: 80, center:[true,true,false]
}),
			translate([0,0,params.b1-f],cylinder({d1: params.d2,d2: params.d2-(f*0.589), h: f, fn: 80, center:[true,true,false]})),
			cylinder({d: params.d3, h: params.b2, fn: 80, center:[true,true,false]
})),
		rotate([params.a, 0, 0], translate([params.d3/2,(params.b1*tan(params.a))/2,-2-e*2],cube({size: [params.d3, params.b3, params.b1*1.3 + 4], fn: 80, center:[true,true,false]}))),
		translate([0,0,-1],cylinder({d: params.d1, h: params.b1+2, fn: 80, center:[true,true,false]})))
   ];
}
```
Many Thanks

Tom

joostn, 2017-01-17:
Hi Tom,

OpenJsCad.org has added global operators, these are not present in the originaljoostn.github.io OpenJsCad. 

The supported syntax is documented here:
http://joostn.github.io/OpenJsCad/index.html

closed 2017-01-17 by joostn

TommyK82, 2017-01-17:
Hi Joostn,

many thanks I didn't realized that .... I changed  it and now it is working.

Thanks

Tom
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

  
    

    
    
    
    
      http://joostn.github.io/OpenJsCad/index.html